kylecorry31/Trail-Sense #2336: Use sentence casing in user guides

kylecorry31, 2024-04-20:
All titles and references to settings / tools should be using sentence casing.

muryno, 2024-04-22:
Are those to be sentence case also or remain title case 

All tool names
Weather Monitor
all quickActions
User guide
Cliff Height
Lightning Strike Distance
Metal detector
White Noise

kylecorry31, 2024-04-22:
All of those should be left as title case for now (and any other tool)

muryno, 2024-04-23:
If  User guides (all headings and ensure the references to text in the UI is still correct) is part of what needs to be converted to Sentence case, 

Are there any exception on those.... 
Also thinking, if some needs to stay Title case, wouldn't that look odd as there is no consistency 😀

Photo Maps
 Triangulate Locations
Bubble Level
Water Biold Timer
Light Meter
Solar Panel Aligner
Light Strike Distance
Augumented Reality
Metal Detector
Packing Lists
QR Code Scanner
White Noise
Recommended apps

kylecorry31, 2024-04-23:
> If User guides (all headings and ensure the references to text in the UI is still correct) is part of what needs to be converted to Sentence case,
>
> Are there any exception on those.... Also [...]

Hi, yes for now keep the tool names and quick action names in Title Case in the User Guide - if I decide to make those in sentence case, I will update the guides as well (essentially anywhere in the app that the tool name appears it should be in title case). I'm on the fence on whether those are considered Product/Branded names and haven't come to a decision yet.

kylecorry31, 2024-04-24:
After further consideration, tool and quick action names should remain in Title Case (how they are today). I may change my mind on some of the quick actions, but for now to maintain consistency, they should not be changed as part of this issue.

muryno, 2024-04-26:
does it look good now or still needed an update

kylecorry31, 2024-04-26:
Hi, going tool to tool, I'm still seeing several strings which are in title case, and also the user guides still need to be converted.

Let me know what you plan to do and I can do the rest

muryno, 2024-04-26:
user guide has tool name which was advices to be in title case....

ill update soon and you can review again

kylecorry31, 2024-04-26:
Yes mentions of tool names in the user guide should remain in title case. 

For reference, when I say user guides I mean the content of the guides themselves - not the list where you select the guide to view

kylecorry31, 2024-05-12:
In preparation of the release next weekend, I'm going to be making these changes.

@muryno if you have any in progress changes that you want included, I ask that you submit a pull request in the next day or two

closed 2024-05-17 by kylecorry31