kylecorry31/Trail-Sense #2362: Metal detector sound icon missing

kuba-orlik, 2024-05-01:
On some phones the "sound" icon is present for the metal detector (like on pixel 7a with lineage), but on pixel 7 pro with stock android the icon is missing
[image]
[image]

kylecorry31, 2024-05-01:
Hi, what happens if you open the Whistle tool on the device where the icon does not show up? Does it get an error message?

Can you also confirm that both are running the same version of Trail Sense (Tools > Settings > Version)?

kuba-orlik, 2024-05-01:
The whistle works fine on the device where the icon doesn't show up - it's loud and clear, no errors. 

In fact the one on pixel 8 was on 5.9.0, not on 5.10.0. Updating the repositories from fdroid helped Apologies for the confusion :pray:

closed 2024-05-01 by kuba-orlik