kylecorry31/Trail-Sense #2405: Packing List Feature Ideas

CallamJ, 2024-05-28:
## Some general ideas for the packing list feature.

## Ways to import/export packing lists.
**This could include:**
- Copying a packing list to clipboard as plain text,
- Exporting a packing list to a text file as plain text.
- Exporting a packing list to an importable (possibly human readable and copy/paste safe text) file. If this was a text file copy to clipboard would be handy.

These could include or not include the current or desired packed amounts.

A feature to quickly increment or decrement packed amounts by 1.
This could be done with an up and down arrow on each item. 
Tapping the corresponding arrow would increment the item's packed amounts up/down by 1.

kylecorry31, 2024-05-28:
Thank you for the suggestions, I will add all of these (and will likely split this issue into smaller issues).

kylecorry31, 2024-05-29:
Updated #1149 to include copy to clipboard and added https://github.com/kylecorry31/Trail-Sense/issues/2406 for the increment decrement buttons.

closed 2024-05-29 by kylecorry31