kylecorry31/Trail-Sense #2435: Compass status in Astronomy 3D or Augmented Reality

astewart77, 2024-06-16:
Compass status in the lower left corner is "Good" in the Navigation tool and Diagnostics are all good.

In Astronomy 3D or Augmented Reality it is (almost) always "Unknown". While trying setting changing Sensors->Compass->Source to various options, once with Mag+Gyro (Custom) the status became "Good". After changing the option back and forth to see if that was causing the problem, I can't get "Good" again for any option.

Everything seems to be functional in Augmented Reality even with the "Unknown" compass - overlays line up with reality and move correctly,  and I can always switch to the Navigation tool and see a "Good" compass status.

Running Trail Sense 6.0.0 on Android 14 / Kyocera Duraforce Pro 3 phone. The symptom was also present before with Trail Sense 5.10.0 on Android 13.

kylecorry31, 2024-06-16:
I'll take a look at this tomorrow, I'm able to reproduce this with the options other than Custom it seems.

kylecorry31, 2024-06-16:
The fix will be in the next release

closed 2024-06-16 by kylecorry31

astewart77, 2024-06-22:
6.1.0 fix confirmed good for me. Thanks.

commit 18e7eaf6 (2024-06-16): Use magnetometer quality for orientation sensors

Closes #2435