kylecorry31/Trail-Sense #2489: Add active map button to photo maps tool

kylecorry31, 2024-07-20:
Either just the Open Photo Map quick action as a button on the photo maps list, or a special list item for the active map (shows map name with an indicator that this is likely the one you are looking for).

michaelblyons, 2024-07-21:
This is the \[On Map\] badge, but for photo maps that are nested inside folders?

kylecorry31, 2024-07-21:
> This is the [On Map] badge, but for photo maps that are nested inside folders?

I was thinking more like the Open Photo Map quick action - it would always show the same map, which is determined by your current location and destination beacon. 

So I guess it would be similar to that (except only one would be chosen), and more useful when you have maps inside folders.