kylecorry31/Trail-Sense #2503: [Bug] Scroll reset when going back after selecting tool

danielwerg, 2024-07-25:
## Issue

After [recent update](https://github.com/kylecorry31/Trail-Sense/releases/tag/6.2.1) I decide to read user guide for few tools and then try to use it but scroll position would keep getting reset to 0 and I would have to scroll down and find last tool I checked on repeat.

## Expected behavior

Remember scroll position for tools tab

## Steps to reproduce

1. Switch to "Tool" tab
2. Scroll down any amount
3. Select any tool
4. Press back button
5. Scroll position reset to 0

kylecorry31, 2024-07-25:
@danielwerg, thank you for the detailed report. This is a known issue, but I can bump it up a little in priority: https://github.com/kylecorry31/Trail-Sense/issues/2061

closed 2024-07-25 by kylecorry31

danielwerg, 2024-07-25:
oh my bad, idk how I didn't see #2061

kylecorry31, 2024-07-25:
No worries, GitHub search only uses keyword matching (from what I can tell), so it can sometimes be tricky to find existing issues, especially when there are so many on the backlog.