kylecorry31/Trail-Sense #2505: Draw GPS heading on mapviews

kylecorry31, 2024-07-25:
Not sure if this should be enabled by default, but draw a line from the user's position to indicate which way they are moving on the mapview (photo maps, paths, navigation/radar compass).

This could help if the compass isn't calibrated but the user is moving.

maverick74, 2025-12-26:
Installed TrailSense on a recent phone!!!

To my most amazingness the thing **does not have a compass** 😠😭🙁😱 

How is this possible?! I though this was even more basic stuff than Bluetooth!!!!!!!!!!! Dagnabit!!!

Well... stopping with the complains: how viable is using GPS/Galileo as an alternative to (x-men's) "Magneto" sensor???

kylecorry31, 2025-12-26:
> Installed TrailSense on a recent phone!!!
>
> To my most amazingness the thing **does not have a compass** 😠😭🙁😱
>
> How is this possible?! I though this was even more basic stuff than Bluetooth!!!!!!!!!!! [...]

Unfortunately GPS is not a great alternative since it only provides an accurate heading while you are moving. I will plan on implementing this feature soon, since I'm focusing on map layers in 2026.

maverick74, 2025-12-27:
Does Galileo bring anything else over GPS in this subject?

I really hate to see a great app like Trail Sense being limited... it was one of the first apps I installed and I was really shocked with the message!

Thank you for the attention 🙂

maverick74, 2025-12-27:
> It's always surprising when manufacturers don't include a compass sensor these days since it is really useful for maps applications.

It was an absolute surprise!!! Not having a magnetic sensor is like lacking wireless or Bluetooth...

Wt...?! Who removes this kind of things??? Is it possible to be THAT DUMB?!.......

Well... it's done...

Thank you for your time in clarifying things to me!

I'm looking forward to this issue to get some limited features