kylecorry31/Trail-Sense #2510: Lunitidal Intervals

kylecorry31, 2024-07-28:
Investigate lunitidal intervals and see if that improves accuracy of the tides tool. If so, add an option to use the calculated lunitidal interval (requires a high tide and a location)

kylecorry31, 2024-07-29:
Initial investigation shows this can improve the tide prediction by just a little bit (on the order of minutes) - but it would be a good step forward to the harmonic tides

kylecorry31, 2024-08-03:
#2519 will add the ability to use custom times

closed 2024-08-03 by kylecorry31

leavenode, 2024-09-02:
I see the new option for lunitidal interval, but I can't figure out how to use an already known value to create a new tide chart? It should be as simple as 
'time of high tide' = 'time of the local upper or lower culmination of the moon' + 'local lunitidal interval'
Am I missing something obvious here?

kylecorry31, 2024-09-02:
Hi, the lunitidal interval that is currently implemented requires you to enter a high tide and it will be automatically estimated. #2519 will add the feature you are looking for.