kylecorry31/Trail-Sense #2514: Position is usually unavailable/"Stale"

NicolaSmaniotto, 2024-07-29:
I'm not sure how to report this, but Trail Sense has a hard time determining my position.

I've tried this multiple times. In the same spot Organic Maps can find it very quickly with good precision (both with "High accuracy" and "Device only"), while Trail Sense almost never finds it at all. Sometimes it finds it after a long wait, but only outside and far from buildings and trees. The fact that OM has already found it does not make any difference.

Android tells me the app is "using my location", but the Sensor Details page shows 0 satellites.

Phone: Oneplus X
Android version: 10
Trail Sense version: 6.2.1

kylecorry31, 2024-07-29:
Hi, I'm sorry you are having issues with Trail Sense. Can you provide me the following information?

1. Are you using a custom ROM?
2. When did this issue start?
3. Does it improve if you disable "Require satellite fix" under Trail Sense Settings > Sensors > GPS? It could be that your phone is not reporting satellite counts
4. Does Trail Sense report any diagnostic issues? (Tools > Diagnostics)

NicolaSmaniotto, 2024-07-29:
Hi, thanks for the quick response.

1. Yes, LineageOS 17.1 (unofficial)
2. Since I installed the app some months ago (F-Droid), don't remember the version
3. It does not
4. Sometimes it reports "GPS signal lost", but not always

kylecorry31, 2024-07-29:
> Hi, thanks for the quick response.
> 
> 1. Yes, LineageOS 17.1 (unofficial)
> 2. Since I installed the app some months ago (F-Droid), don't remember the version
> 3. It does not
> 4. Sometimes it reports "GPS signal lost", but not always
> 
> EDIT: I installed it at the end of March, so it was probably 5.9.0

Thank you for the clarification. I'll do some tests on a phone with LineageOS in case it is an issue with the ROM (though it could be phone specific).

While I don't think this will be the problem, it may help to clear the location cache (Trail Sense Settings > Sensors > GPS > Clear cache). But the fact that it says 0 satellites makes me think that is not the case (normally disabling "require satellite fix" resolves this issue).

kylecorry31, 2024-08-03:
@NicolaSmaniotto I made some changes that I think might help when the "Requires satellite fix" setting is disabled. This will be in the next release, but it is also available in the latest Nightly build if you wanted to try it out (it installs as a separate app, so all your data will remain intact if you install it). Instructions on how to download and install the Nightly builds can be found here: https://github.com/kylecorry31/Trail-Sense/discussions/1940

If you do try out the Nightly build, please let me know if that resolves this issue for you.

closed 2024-08-09 by kylecorry31

NicolaSmaniotto, 2024-08-10:
Sorry for the late reply, I was away for the week. The nightly build behaves the same as the released one.

I think I've found the reason why Organic Maps is so much faster when trying to find the position. It is apparently using [A-GPS](https://en.wikipedia.org/wiki/Assisted_GNSS), which greatly reduces the time needed in the beginning. Indeed, if I put the phone in airplane mode, OM takes a long time to find the position (just like TS).

Have you considered a similar strategy? I understand you don't want to have features that _require_ internet access, but in this case it would be speeding up the GPS resolution by using additional data, when such internet access is available.

kylecorry31, 2024-08-10:
Thank you for getting back to me. In this case, it appears that your phone itself is delayed in getting a GPS response - not sure if that's phone hardware, an OS issue, or just lack of strong satellite signals in your location. 

I have considered A-GPS, and you can see the output of my research here: https://github.com/kylecorry31/Trail-Sense/issues/432 and https://github.com/kylecorry31/Trail-Sense/issues/972 . I may consider reopening 432 though - in your area does the A-GPS get good accuracy while hiking? The main reason I closed it because in my testing it was very inaccurate for most use cases in TS, and I believed waiting for GPS would be better.