kylecorry31/Trail-Sense #2529: Improve frequency abbreviation for backtrack

kylecorry31, 2024-08-07:
30m is intended to be 30 minutes, but it can just as easily be misinterpreted as 30 meters. Look into displaying 30 min or 30 minutes if no seconds are provided.

astewart77, 2024-08-08:
That raises a useful feature request - Backtrack interval by distance delta. If you are moving at various speeds or standing still for a while, you get a nicely spaced track. I turned on my old Garmin and I had it set to 100 feet.

kylecorry31, 2024-08-08:
> That raises a useful feature request - Backtrack interval by distance delta. If you are moving at various speeds or standing still for a while, you get a nicely spaced track. I turned on my old Garmin and I had it set to 100 feet.

Thank you for the suggestion, I have created https://github.com/kylecorry31/Trail-Sense/issues/2532 to track that