kylecorry31/Trail-Sense #2532: Backtrack record every X distance

kylecorry31, 2024-08-08:
Rather than recording by time, allow the user to record every ~X distance. This can be done using either the pedometer or with #1719 (speed).

astewart77, 2024-08-09:
I like the pedometer idea better than actual feet/meters. I could set Backtrack to my preferred number of steps and leave it that way. Then calibrate the pedometer at the beginning of a hike for the effect of terrain/weather/pack weight on me.

kylecorry31, 2026-10-11:
This may be tricky to implement - pedometer is not a wake up sensor, so the device can enter doze mode if neither the pedometer or backtrack is holding a wakelock. If backtrack is holding a wakelock then this can work at the expense of battery life (though somewhat minor from my experimentation), as the pedometer can continue to update in real time. I can display a notice to users that the device won't enter a sleep state while this is active.