kylecorry31/Trail-Sense #2537: Breathing guide

kylecorry31, 2024-08-20:
Suggestion was the wim-hof breathing method, which can help with cold exposure or maybe altitude sickness.

Links:
* https://inner-breeze.app/
* https://www.healthline.com/health/wim-hof-method

kylecorry31, 2024-09-02:
Merging this into the survival guide (#2539)

jm355, 2024-09-02:
Just chiming in, it's really useful to have an app actually walking you through the breathing like the inner breeze app does, it makes it a lot easier to understand than a text description of the technique does, especially when you're in the middle of the exercise and might have lost count of how many breaths you've done in the in-out phase, or in general if you've lost track of how many rounds you've done

kylecorry31, 2024-09-02:
Thank you for the suggestion, I will look into that as well

kylecorry31, 2026-08-24:
I'm going to keep this as part of the survival guide only and if anyone has an app recommendation they can post on https://github.com/kylecorry31/Trail-Sense/discussions/4002

closed 2026-08-24 by kylecorry31