kylecorry31/Trail-Sense #2539: Survival guide (research)

kylecorry31, 2024-08-20:
See if there are any survival guides compatible with Trail Sense's license that can be embedded into the app and don't take up a lot of storage space. Maybe a text version of https://irp.fas.org/doddir/army/fm3-05-70.pdf, which is public domain.

kylecorry31, 2024-08-22:
Going with https://armypubs.army.mil/epubs/DR_pubs/DR_a/pdf/web/ARN12086_ATP%203-50x21%20FINAL%20WEB%202.pdf and I am almost certainly going to implement this.

jm355, 2024-08-23:
https://github.com/ligi/SurvivalManual/issues/206#issuecomment-903144379 in this comment @ligi says the content of the app is public domain. So maybe you can use it after all?

kylecorry31, 2024-08-23:
> [ligi/SurvivalManual#206 (comment)](https://github.com/ligi/SurvivalManual/issues/206#issuecomment-903144379) in this comment @ligi says the content of the app is public domain. So maybe you can use it after all?

Great find! I will look into that and compare it against the work I've been doing to embed the Army guide.

kylecorry31, 2024-08-26:
FYI, I've decided to write my own survival guide based on the Army guide and my own knowledge (fact checking everything against credible sources, mainly the Army guide). My rationale is the existing guides are very in depth (Army and the Survival Manual) - which is great if you want to learn a lot - but they can also be a bit hard to decipher in a survival situation. I've rewritten a couple of chapters (reducing them by about 75 - 80%) to focus on the key information and it didn't take me too long (though I did start with the shorter chapters).

It will probably take a few passes to get them easy to read while still being useful, but I think it will be worth it in the long run.

closed 2024-10-12 by kylecorry31