kylecorry31/Trail-Sense #2562: Solar panel aligner sometimes gives very incorrect azimuth

Stormwind99, 2024-09-18:
Solar panel aligner sometimes gives very incorrect - like 90 degrees off.

I make sure GPS and Magnetometer are good, and that compass reading makes sense.

[image]
[image]
[image]
[image]
[image]
[image]

P.S. Thanks for Trail Sense! I use it frequently when camping (full-time now), hiking, and backpacking.

kylecorry31, 2024-09-18:
Thank you for the detailed report, it definitely seems like the azimuth is incorrect in this case (the target azimuth, not the current device azimuth). I'm able to reproduce the issue on my device as well.

I'll look to fix this in the next release.

kylecorry31, 2024-09-18:
This should be fixed in the nightly build (https://github.com/kylecorry31/Trail-Sense/discussions/1940). 

The next release will be in early - mid October.

closed 2024-09-18 by kylecorry31