kylecorry31/Trail-Sense #2563: Offline survival manual Recommended App link broken

Stormwind99, 2024-09-19:
Extremely minor bug: 

User Guide -> Recommended Apps -> Offline survival manual -> Download link leads to missing Google Play Store item "Item not found"

Might be deprecated since I think I read you are working on a built-in survival manual.

kylecorry31, 2024-09-19:
It looks like it is no longer available on Google Play. I assume because a new release hasn't been published in a long time, it does not meet the Play policy requirements 

I'll see if I can link to the F-Droid version directly.

kylecorry31, 2024-09-19:
Looks like the only thing preventing the Survival Manual from releasing is broken tests. I'm going to try to fix them.

kylecorry31, 2024-09-19:
Hopefully this will allow the Survival Manual to be republished on Google Play: https://github.com/ligi/SurvivalManual/pull/290

kylecorry31, 2024-09-26:
It is back on Google Play and I suspect a new release will happen soon

closed 2024-09-26 by kylecorry31