kylecorry31/Trail-Sense #2580: Publish release build on GitHub releases

kylecorry31, 2024-10-07:
Has better performance and smaller APK size than the debug build.

Keep posting the debug build in addition.

ticklemyIP, 2025-03-07:
Hi, I'd just like to ask whether the current "dev" versions of TrailSense are any different than the Play and F-droid versions other than the signing keys? Can the "dev" version be considered a full fledge GitHub release?

kylecorry31, 2025-03-07:
> Hi, I'd just like to ask whether the current "dev" versions of TrailSense are any different than the Play and F-droid versions other than the signing keys? Can the "dev" version be considered a full fledge GitHub release?

Hi, the dev version has full debug support enabled, which increases the APK size, makes it a bit slower, but allows you to fully debug errors in the app. It also contains the following extra features for debugging: https://github.com/kylecorry31/Trail-Sense?tab=readme-ov-file#debug-features

This is considered a full GitHub release, and this issue is to add a second GitHub release with the optimized release APK for those who don't plan on debugging and just want the APK as early as possible.

ticklemyIP, 2025-03-07:
Thanks for the reply, that clears things up. Is there an estime about when we can expect regular releases?

kylecorry31, 2025-03-07:
I can take a look at getting this ready for the next release because I think it may be easy to do, but no gaurantee. 

(release will be in a couple of weeks)

kylecorry31, 2025-03-07:
This should be all set for the next release - a second APK will be present on the release assets (*-github-release.apk).

closed 2025-03-07 by kylecorry31