kylecorry31/Trail-Sense #2587: Host GPX files on website with key points

kylecorry31, 2024-10-18:
Should allow user to specify which region to download.

These can be hosted on the Trail Sense website.

- [ ] Mountain peaks (major)
- [ ] Mountain peaks (all)

Community contribution is welcome. All contributions should be in the GPX file format as waypoints. It should include latitude, longitude, and a name (in English for now).

adryzz, 2024-10-20:
I feel like this task can be accomplished using [Overpass Turbo](https://overpass-turbo.eu/)

i'm not particularly skilled with it, and i'm sure they can be narrowed down, but ALL mountain peaks sounds like a TON of data.

For example, this query shows all mountain peaks > 2000m in elevation.
[image]

([GPX export of the query](https://github.com/user-attachments/files/17449808/export.txt), 4.5MB)

kylecorry31, 2024-10-20:
> I feel like this task can be accomplished using [Overpass Turbo](https://overpass-turbo.eu/)
> 
> i'm not particularly skilled with it, and i'm sure they can be narrowed down, but ALL mountain peaks sounds like a TON of data.
> 
> For example, this query shows all mountain peaks > 2000m in elevation. ![screenshot of overpass turbo showing thousands of results in the alps](https://private-user-images.githubusercontent.com/46694241/378175855-1ba1b893-c3d5-4b70-9be6-d3a8aa619625.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk0MjAzNjEsIm5iZiI6MTcyOTQyMDA2MSwicGF0aCI6Ii80NjY5NDI0MS8zNzgxNzU4NTUtMWJhMWI4OTMtYzNkNS00YjcwLTliZTYtZDNhOGFhNjE5NjI1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIwVDEwMjc0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUxYWZhODY2MWY0NWJlYmViYTcwZDNmZGUzMzcwMDcwYjhkNDdjNjdiYjlhZThmYzQ2OTc2NzliNDcwMmU1YmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.btOA8VM4C14u0XSl7mQGnkXgLYdGgemhP0KrS4pjE4c)
> 
> ([GPX export of the query](https://github.com/user-attachments/files/17449808/export.txt), 4.5MB)

That's a really cool tool, thank you for pointing this out

kylecorry31, 2024-10-20:
I'll look into providing a simple wrapper around that api on the website

kylecorry31, 2024-10-20:
I added that query to the user guide for beacons and modified this item to be about assembling a wrapper around it to host on the Trail Sense website