kylecorry31/Trail-Sense #2590: New Feature Ideas

kylecorry31, 2024-10-20:
This is a continuation of #1911, since that was getting too long.

Please feel free to post new feature ideas in the comments on this issue. I would love to have some other thoughts about what should be included in this app (no guarantee that I will include them, but I will consider each idea). No idea is a bad idea here, worse case I may say that I will not include it in Trail Sense. Some features may be better suited as plugins or a different app. Plugins will allow the usage of Internet when available, but must have offline support.

Some set in stone goals of this app:
- Trail Sense must not use the Internet in any way, as I want the entire app usable when there is no Internet connection
- New features must provide some benefits to people using the app while hiking, in a survival situation, etc. 
- New features must be based on peer-reviewed science
- New features must not increase the app size much (available size is dependent on feature importance). App should remain under 10 MB.

For reference, some things I have considered in the past and removed were:
- Animal behavior
    - I can't say there is enough science / supporting evidence behind Solunar theory for me to make accurate predictions of animal behavior in general, though if someone provides strong evidence I will consider it
- Heart rate sensor
    - I don't want Trail Sense to move into the medical sector, there are increased regulations on that in Google Play and not something I'm willing to provide support for.

RAPXT, 2024-10-21:
For phones that make use of auto dark mode and which have an amoled screen, it would be nice to have a "system black" theme.

Which would save noticeably more battery compared to the dark grey the current (single) system theme uses.

RAPXT, 2024-10-21:
I mean the system wide theme changer of android, which changes themes of apps by sunset/sunrise or custom time.

But thats not the important part, i wanted to say, that i miss a theme, that enables when the system requests a dark theme. Addtionally to the current system theme, there should be a "system black" theme, which changes the background to black instead of darkish grey.

[image]

Simplest thing to notice the difference is in Trail Sense to change to system theme and then black theme. I would like a system theme with that black background.

kylecorry31, 2024-10-21:
> I mean the system wide theme changer of android, which changes themes of apps by sunset/sunrise or custom time.
> 
> But thats not the important part, i wanted to say, that i miss a theme, that enables when the system requests a dark theme. Addtionally to the current system theme, there should be a "system black" theme, which changes the background to black instead of darkish grey.
> 
> ![Screenshot_20241021-161911040](https://private-user-images.githubusercontent.com/53192319/378471142-49ca4643-f4e2-4640-93cd-e5f9a49b3e3e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk1NTEwNjIsIm5iZiI6MTcyOTU1MDc2MiwicGF0aCI6Ii81MzE5MjMxOS8zNzg0NzExNDItNDljYTQ2NDMtZjRlMi00NjQwLTkzY2QtZTVmOWE0OWIzZTNlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIxVDIyNDYwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFjYWUzNWM1NzEwMWY3MWIzZDQyYWYyMWMxNzQ1MWMzNzhmY2MzNzJhN2NmM2IxMjI4NTAyMWNhOGFlM2Q1YmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.NgJ-VJUpRPlEf24h-CkwPDk-OIM4gbc2SzpeILkgwqk)
> 
> Simplest thing to notice the difference is in Trail Sense to change to system theme and then black theme. I would like a system theme with that black background.

Thank you, I see - I was not aware that was a feature on Android. Here's the issue to track that: https://github.com/kylecorry31/Trail-Sense/issues/2592

damianofalcioni, 2024-10-23:
Hi,
I know the feature idea ist is quite long but please consider the integration with brouter. it is a famous open source app able to generate paths, totally offline.
It can be really helpful to automatically create a path from your position to a beacon or between multiple beacons (useful also as a backup for the backtrack feature). Additionally it generate also turn by turn instructions that could be helpful to send as notifications while following path.
Thanks

kylecorry31, 2024-10-23:
> Hi, I know the new feature ideas list is quite long but please consider the integration with brouter. it is a famous open source app able to generate paths, totally offline. It can be really helpful to automatically create a path from your position to a beacon or between multiple beacons (useful also as a backup for the backtrack feature). Additionally it generate also turn by turn instructions that could be helpful to send as notifications while following path. Thanks

Thank you for the suggestion, I've created this issue to track the research into this: https://github.com/kylecorry31/Trail-Sense/issues/2595

7eventech77, 2024-10-30:
List of ideas I currently have at the moment. Some of these may not be possible due to me not knowing how things work.

Feature Requests
- Ability to calibrate Trail Sense compass with another physical known accurate compass.
- Ability to use yaw axis of gyroscope as a short term compass when aligned with a physical compass
- Ability to calibrate compass with Sun or moon through augmented reality
- Display current temperature reading on navigation screen
- Use Ambient Light sensor as potential trigger for trail sense theme

kylecorry31, 2024-10-30:
> List of ideas I currently have at the moment. Some of these may not be possible due to me not knowing how things work.
>
> Feature Requests
>
> * Ability to calibrate Trail Sense compass with another [...]

Thank you for the suggestions:

- Here's an issue to track the compass calibration. https://github.com/kylecorry31/Trail-Sense/issues/2603 I put a note in there about why it wasn't saved in the past. 
- For using the AR calibration mode elsewhere, see this issue: https://github.com/kylecorry31/Trail-Sense/issues/2419
- I do not have plans to add the current temperature to the navigation screen, because I would like to keep those tools separate. It is displayed on the weather tool and will eventually be displayed on the climate tool (#1904). I also have plans make the weather tool widely available if you device does not have a barometer (#2090).
- Unfortunately, due to the way theming works on the Android framework that TS uses, it needs to be set on load. If the light sensor is used, there will be a noticeable delay when opening TS. I'll investigate it though: https://github.com/kylecorry31/Trail-Sense/issues/2604

kylecorry31, 2024-10-30:
> Hello, would you be willing to add millimeters, teaspoons, and tablespoons to the unit converter? I appreciate the desire for simplicity, there are too many units too capture them all, but I think those are common enough to be worth adding

Thank you for the suggestion, here's a ticket to track that: https://github.com/kylecorry31/Trail-Sense/issues/2605

acarv, 2024-12-29:
As Android 5.1.1 is the last acceptable version of Android (for me in terms of privacy, functionality and usage of reliable and controllable equipment : Galaxy Note 4). As it is also the last version where you can have a minimum control over the OS.
Would You consider to produce a version for it?
Thank You.

kylecorry31, 2024-12-29:
> As Android 5.1.1 is the last acceptable version of Android (for me in terms of privacy, functionality and usage of reliable and controllable equipment : Galaxy Note 4). As it is also the last [...]

Hi, thank you for the suggestion. I have looked into Android 5 support before, but there was too much effort involved to port Trail Sense over to it.

I1138, 2024-12-30:
New user, so I apologize if this is just a usage / settings error. I'd like to be able to turn the GPS on for a few seconds or until the location provider reaches a specified accuracy level before specifying a beacon / photo map calibration point. OTOH this may be limited use for regular hiking. My personal use case is accurately marking survey pins such that I can find them again, which is much easier if I can mark the point in a 10ft radius rather than 100ft. I could see this also being useful for hiking with photo maps when the calibration points are close together.

kylecorry31, 2024-12-30:
> New user, so I apologize if this is just a usage / settings error. I'd like to be able to turn the GPS on for a few seconds or until the location provider reaches a specified accuracy level before [...]

Thank you for the suggestion, I believe the feature(s) you are looking for will be implemented by these:
- https://github.com/kylecorry31/Trail-Sense/issues/2464
- https://github.com/kylecorry31/Trail-Sense/issues/2461

darrencoen, 2025-01-09:
Firstly, trail-sense app is extremely cool and well made, thank you.

On the bubble level I was wondering if you could put the value of the hypotenuse in the 3rd orange circle. I looked at the code a bit to see if I could add it myself but it wasn't obvious to me. It seems like you are already calculating the value of it to place the bubble, but I don't see how to make the text label, nor ensure that it moves with the bubble. (I don't code kotlin/make apps)

[image]

In this example you're 22 degrees off in one direction and 45 in another. But the phone itself is on a ~50 degree plane and i think that's what that bubble should show.

~50 = √(22² + 45²)

It might also make sense to add it with the single decimal point values? What to label it though -- I don't know. It's not z. 

Thanks again!

kylecorry31, 2025-01-09:
> Firstly, trail-sense app is extremely cool and well made, thank you.
>
> On the bubble level I was wondering if you could put the value of the hypotenuse in the 3rd orange circle. I looked at the code [...]

Thank you for the suggestion, I've created this issue to track it: https://github.com/kylecorry31/Trail-Sense/issues/2767

QuantumDamage, 2025-02-20:
Hi, I love your app and I think I might have a feature idea that might fit in it well.
Striking clock (https://en.wikipedia.org/wiki/Striking_clock) - my city have one, but I move further away from it and I can't hear it now ;-(

It would fit in Tools/Clock. Every quarter it would emit quarter sound and additionally in every hour it would emit hour sound. Properly multiplied. 
Example:

- 13:15 - "bim" - since it is quarter past.
- 13:30 - "bim", "bim" - since it is two quarter past.
- 13:45 - "bim", "bim", "bim" - since it is three quarters.
- 14:00 - "bim", "bim", "bim", "bim", "bam", "bam" - since it is full hour, 2 pm.

Possible configuration options:
 - enable
 - start time - for example default 6:00
 - stop time - for example default 22:00
 - "bim" interval - 15, 30 or disable. 15 for default.
 - "bam" - exact, single or disable. exact (2 bams for 2 am/pm) for default.
 - enable in silent mode - enabled by default. Use case: I have the phone constantly in silent mode, but I like to hear alarms, timers and I would like to hear this feature. But some people might like to don't hear this if they are entering silent mode in theaters etc.
 - loudness adjustment
 - mix or interrupt current audio - I observed different behaviors for sounds in Android. Some sounds are mixed with currently played audio and some are pausing it, emitting and returning to playback. It would be nice if there would be an option to define it if possible.

Use cases:

- Sometimes I'm doing something absorbing, and I'm loosing track of time. It would be nice to hear simple audio clues what time it is. Manually setting and unsetting alarms is tedious. This feature would help to track time without interruptions and touching phone.
- Visually impaired people could also benefit from it.

Risks:
- I'm not sure if this is the case, but it might use some additional energy to track alarm for every quarter.
- It might consume some energy to emit sound. Not sure how much tough.
- bim and bam sounds would add some kilobytes to binary.

Keep on great work, and thanks for what you already did :-)

kylecorry31, 2025-02-20:
> Hi, I love your app and I think I might have a feature idea that might fit in it well. Striking clock (https://en.wikipedia.org/wiki/Striking_clock) - my city have one, but I move further away from [...]

Thank you for the suggestion and excellent write up. I'll plan to add that: https://github.com/kylecorry31/Trail-Sense/issues/2826

notknowcode, 2025-02-23:
hi. thanks for the good app. there are a few suggestions: 
1. in the tab with all utilities, when you open any and go back, the top of the list opens, not the previous place. 
2. in viewing the sun through the camera, the ability to change the time of day, this is very useful in tourism, to know how best to put up a tent without the morning sun.

kylecorry31, 2025-02-23:
> hi. thanks for the good app. there are a few suggestions:
>
> 1. in the tab with all utilities, when you open any and go back, the top of the list opens, not the previous place.
> 2. in viewing the sun [...]

Thank you for the suggestions. I have plans to add both of those:
- https://github.com/kylecorry31/Trail-Sense/issues/2061
- https://github.com/kylecorry31/Trail-Sense/issues/2222

kylecorry31, 2025-03-04:
> Hi, I don't know if it would quite in line with the app's focus, but I wanted to suggest the idea of calculating golden hour and blue hour - either as part of the astronomy tool or as a separate tool within the app. Their time is based on the sun's position, and it'll vary depending on location and season, but since it can already calculate the sun's position it might be possible to have it piggy back off that feature while remaining small & offline.
> 
> Have a good one

Thank you for the suggestion, unfortunately I do think that is better suited for a photography app rather than Trail Sense.

jm355, 2025-03-15:
Hello, would you be willing to add milligrams to the unit converter?

kylecorry31, 2025-03-15:
> Hello, would you be willing to add milligrams to the unit converter?

Thank you, yes I can add that. Here's an issue to track: https://github.com/kylecorry31/Trail-Sense/issues/2851

kylecorry31, 2025-04-04:
I'm looking for additional field guide page suggestions. If you would like to get a page added, please read through the following discussion post and leave a reply there: https://github.com/kylecorry31/Trail-Sense/discussions/2876

Thank you!

peter9421, 2025-04-26:
Some of my favorites for new feature ideas.

- Option to keep display on at least while in map perhaps /w battery % limit.

- Selectable TTS announcements based on set intervals of time or distance.  For example every hour announce current time, time to sunset, ETA, etc. Every change in mile announce distance from/to start or beacon.  If not TTS so long as text e.g. "Current time is 2 PM" is in notification third party apps like voice notify can do the voice announcements.

- In gray area where map name is displayed one or two widget like selections to show bits like current speed, total distance, distance to beacon / start rather than map name.

kylecorry31, 2025-04-26:
> Some of my favorites for new feature ideas.
>
> * Option to keep display on at least while in map perhaps /w battery % limit.
> * Selectable TTS announcements based on set intervals of time or distance. [...]

Thank you for the suggestions!

1. Here's a new issue to track that: https://github.com/kylecorry31/Trail-Sense/issues/2915
2. I do not have plans at the moment to support TTS, but I reopened this issue in case anyone would like to contribute this feature: https://github.com/kylecorry31/Trail-Sense/issues/511
3. Unfortunately, I don't think I'm going to add this one since there would be a lot of work involved to make that possible. If you have backtrack enabled, you can see the total distance in the notification or in the widget panel (long press bottom navigation - you may need to add the widget in settings). I have plans to add a spedometer to the widget panel as well. As for the distance to a beacon, if you are navigating that will show at the bottom of the map.

susukin0, 2025-05-06:
Hello can we make additional alarm system for bubble level? if it moves some degrees for a routine period then the alarm rings. this can be helpful for detecting earthquakes locally... i appreciate the project!

kylecorry31, 2025-05-06:
> Hello can we make additional alarm system for bubble level? if it moves for some very small degrees for a routine period then the alarm rings. this can be helpful for detecting earthquakes locally... i appreciate the project!

Hi, I'm trying to figure out how this would be used in a real world scenario, can you ellaborate?

I'm thinking if you leave your phone on the ground while camping and it detects vibration, either you would already know something is happening, the earthquake would be too small to matter outside, or it would be a false positive (ex. rolled over in the sleeping bag and hit your phone). As of now, I'm not sure that I will include it - it sounds similar to https://github.com/kylecorry31/Trail-Sense/issues/679 which I had previously evaluated.

kylecorry31, 2025-05-07:
> Battery discharge alert
> 
> I have had trouble with unreliable chargers. I plug my phone in, it charges, but then it stops charging later on.
> 
> I wish Trail Sense had some sort of alarm, by ringing or perhaps just an occasional notification sound, that goes off when the battery starts discharging when it's supposed to be charging.

Thank you for the suggestion, I have plans to add this eventually (open to community contribution): https://github.com/kylecorry31/Trail-Sense/issues/2905

amboku-seks, 2025-05-12:
We need Turkish Translate your app is so nice Türkiye Your app is really useful since it is an earthquake country. We shouldn't have to translate in case of emergency.

kylecorry31, 2025-05-12:
> We need Turkish Translate your app is so nice Türkiye Your app is really useful since it is an earthquake country. We shouldn't have to translate in case of emergency.

Hi, translations are community contributed. If you would like to see more Turkish translations, you can find the links to contribute in this issue: https://github.com/kylecorry31/Trail-Sense/issues/213

Thanks!

user44000, 2025-05-12:
Quick Setting tile for the compass. For using the compass from the lockscreen.

kylecorry31, 2025-05-12:
> Quick Setting tile for the compass. For using the compass from the lockscreen.

Hi, can you provide more information about what feature you are looking to get added? Unfortunately, I'm not following.

user44000, 2025-05-13:
Like this app. Then people wouldn't need a separate app for this feature.
https://github.com/Tetr4/CompassTile

kylecorry31, 2025-05-13:
> https://github.com/Tetr4/CompassTile

Thank you, that is pretty neat. Here's an issue to track that: https://github.com/kylecorry31/Trail-Sense/issues/2942

realgooseman, 2025-05-16:
Hi @kylecorry31,

Here's a new feature suggestion: an offline worldwide emergency wiki.
I will give you a brief summary and example below, let me know what you think.

It would come in the form of a page (like the Survival Guide) with basic information about the 195 countries (WikiVoyage style but condensed with most important info only). There would be a search bar to search for specific countries and you could also favorite them to the top for easier access. The information could also be extended in the future with more data.

The pages would be text based with nice layout for easier reading, except for the additional country flag images which could come in handy if you see a flag in an emergency situation and don't recognize it. The data for each country can easily be found on the internet and likely won't change in the foreseeable future.

### **Country Wiki Page Example**

Country Flag: [IMAGE]
Country : United States of America
Continent: North America
Spoken language(s): English, Spanish, Hawaiian
Currency: American Dollar / USD / $
Country Code: +1 / 001
ISO Code: US / USA
Police: 911
Ambulance: 911
Fire Department: 911
Poison Center: 1-800-222-1222

### **Useful links to gather the data for this feature:**
Flag/Country/Continent/Spoken Languages/Currency: [WikiVoyage](https://en.wikivoyage.org/wiki/Main_Page)
Country Codes: https://countrycode.org/
Emergency phone numbers: https://worldpopulationreview.com/country-rankings/911-by-country
Poison Center: No documentation but "iFirstAid" or "First Aid" apps on Android/IOS have a nice list of emergency numbers including poison centers.

kylecorry31, 2025-05-16:
> ### **Useful links to gather the data for this feature:**
> Flag/Country/Continent/Spoken Languages/Currency: [WikiVoyage](https://en.wikivoyage.org/wiki/Main_Page) Country Codes: [...]

Thank you, I will plan to add that: https://github.com/kylecorry31/Trail-Sense/issues/2944

I may limit it to just the emergency numbers for now, but will research the other fields as well.

kylecorry31, 2025-05-19:
> 1. If we can set a timer for flashlight and screen light to turn off on its own will be good also if we can adjust the screen light brightness too, a better example is "Lumolight" app on playstore
> 2. If we get a network signal strength alert as i recently on a trip on 3 bars the internet not worked, but as it get 4 or 5 bars the network works fine, so if you can add that too a better example is "Signal Alert" app on playstore

Thank you for the suggestions

1. Here's an issue to track the screen light timer: https://github.com/kylecorry31/Trail-Sense/issues/2948. There's already a timer for the flashlight and a slider to adjust screen brightness. Please see https://kylecorry.com/Trail-Sense/guide?id=tool_flashlight for details on how to use those features.
2. Here's an issue to track signal change alerts: https://github.com/kylecorry31/Trail-Sense/issues/2949

kylecorry31, 2025-05-19:
> It will also be good if we can get a option to stick our network to specific version like LTE only as sometimes moving through remote areas the cell signal switch to 3G even though there are some LTE signals, this won't require much of work as it inside every android phone hidden settings which if added as shortcut a example of it is "5G Only" app on the playstore

Thanks, here's a research issue to track that: https://github.com/kylecorry31/Trail-Sense/issues/2951

kritma, 2025-05-24:
Would be cool if you add support for [OpenStreetMap](https://www.openstreetmap.org/), like downloading/importing, and viewing your location on map.

kylecorry31, 2025-05-24:
> Would be cool if you add support for [OpenStreetMap](https://www.openstreetmap.org/), like downloading/importing, and viewing your location on map.

Thank you for the suggestion, I have the following issues to track that:
- Map tile import support: https://github.com/kylecorry31/Trail-Sense/issues/604
- Downloading (via a plugin, won't be in the base Trail Sense app due to the Internet limitation): https://github.com/kylecorry31/Trail-Sense/issues/1540

stuffpicklesays, 2025-05-29:
It would be really cool to see a WearOS app for trail sense that syncs with your phone. The compass is perfect for a round watch display

kylecorry31, 2025-05-29:
> It would be really cool to see a WearOS app for trail sense that syncs with your phone. The compass is perfect for a round watch display

Thank you for the suggestion, unfortunately I am not currently planning to add WearOS support: https://github.com/kylecorry31/Trail-Sense/issues/1467

phed, 2025-06-22:
I want a QR code generator. Especially for text.

kylecorry31, 2025-06-22:
> I want a QR code generator. Especially for text.

Thank you for the suggestion. You can currently generate QR codes in the notes tool which can be used for sharing data between devices.

993782githbcom, 2025-07-11:
I want a scientific calculator that can replace the work of Google calculator

993782githbcom, 2025-07-11:
Can a module that generates QR codes be added?

993782githbcom, 2025-07-11:
Can you add functions like normal clocks, such as alarm clocks, timers and stopwatches?

kylecorry31, 2025-07-11:
> * The calculator can be used to calculate distances through tight bits and trigonometric functions.
> 
> Functions like secret bits can be provided in the calculator. However, in order to facilitate users' use, users can add formulas by themselves. To prevent users from forgetting the calculation method of formulas, you can label the parameters and names of the formulas. In addition, users can save the calculation history, and some original parameters can also be manually preserved and marked.
> 
> I found that there is no secret function in augmented reality like a military telescope, which can be useful when measuring long-range distances. However, it should be considered that the standards for secret positions between China and the Soviet Union and Europe and the United States are different.

Thank you for the additional context, unfortunately I still believe that falls outside the supported use cases of Trail Sense and I would rather have the calculations built directly into tools. I have plans to add milliradian support (#747)

ElonSatoshi, 2025-07-21:
The compass should notify you if there is metal interfering with the compass, using the same capabilities as the metal detector. The compass quality should at least turn red and show poor. Better yet. it should say "Metal", and/or there should be a red popup saying "Metal interference detected, please move your phone away from anything metal"

kylecorry31, 2025-07-21:
> The compass should notify you if there is metal interfering with the compass, using the same capabilities as the metal detector. The compass quality should at least turn red and show poor. Better [...]

Thank you for the suggestion, I'll plan on adding that. Here's a new issue to track: https://github.com/kylecorry31/Trail-Sense/issues/3062

alphafalcon, 2025-07-26:
Support loading [world files](https://en.wikipedia.org/wiki/World_file) when importing images into offline maps.
I have a workflow to generate map images from opentopomaps for which I already know the coordinates. 

Additionally maybe exporting a world file when the map has been calibrated.

kylecorry31, 2025-07-26:
> Support loading [world files](https://en.wikipedia.org/wiki/World_file) when importing images into offline maps. I have a workflow to generate map images from opentopomaps for which I already know the coordinates.
> 
> Additionally maybe exporting a world file when the map has been calibrated.

Thank you for the suggestion, I've added that to the list in this issue: https://github.com/kylecorry31/Trail-Sense/issues/604

kylecorry31, 2025-07-31:
> Hi, I would like to have a feature on the compass screen that allows me to stay on a bearing from a fixed point.
> 
> Let me explain: I'm an RC model flyer and sometimes a model crashes or gets lost in the bushes or a cornfield. What we do in that case is make sure we remember our current position and then set a bearing as good as we can in the direction of where the model has made ground. Then we try walk along that bearing as straight as we can. Usually the model will not be more then a few meters from that bearing.
> 
> The problem is that following the bearing is quite hard when you are going through bushes or a cornfield. It would be helpful if there was some way of showing how far off the bearing one in the app. I'm thinking along the lines of the ILS 'follow the needle' principle that is used for guiding airplanes on the glidepath to an airstrip. I made a mockup screenshot to show what I mean. The red line is below the compass represents the 'ideal' position exactly on the bearing. The white line indicates how far i'm off the bearing and where I need to go to get back on the bearing. So in this case by walking to the right, I'll get back on the bearing.
> 
> <img alt="Image" width="720" height="1544" src="https://private-user-images.githubusercontent.com/2592283/472667551-f0f28fbb-b6a6-466b-97c3-39218b2bad9c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTM5MjIyNTAsIm5iZiI6MTc1MzkyMTk1MCwicGF0aCI6Ii8yNTkyMjgzLzQ3MjY2NzU1MS1mMGYyOGZiYi1iNmE2LTQ2NmItOTdjMy0zOTIxOGIyYmFkOWMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDczMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA3MzFUMDAzMjMwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzI0MDk0OTczNThhOTlhNGJkM2Q4MjllNDgxZDI3ZDAxODUzNDg1MmZlYTAyMDdhYTI4NzIyOTg5YTU5M2Y4NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.roYhnZo2A0_lIAF_G1vOdESzSkkh_oipGbulY1DOJ0Y">

Thank you for the suggestion. That is a good idea. I think that might be the same as https://github.com/kylecorry31/Trail-Sense/issues/1793 (I believe that's what the Garmin feature is doing). I think your description of that feature is easier to understand, so I'll update that later.

kylecorry31, 2025-08-17:
It's looking like I'll be a bit ahead of schedule with the 2025 feature plans, if there's an issue you would like to see moved up in priority I recommend giving it a "thumbs up" reaction (the main comment in the thread) so that I can sort and see what people most want. Not guaranteed to use that ordering due to effort, feasibility, etc. but it will be factored in. 

I'm leaning toward more map or astronomy features since those are my most used tools. And probably some augmented reality improvements because I like to use that when hiking in the mountains to see what the nearby peaks are.

I won't publish the 2026 feature plan for another couple months, but will start adjusting the priority of issues soon.

forivall, 2025-08-27:
This could be reported as either a bug or a feature: When using the night mode/red filte, the navigation bar is still visible with the system theme, which still disrupts night vision. The easy solution is to add an option to enter full screen when using night mode, and the hard solution is to implement a system filter which changes the color of the entire system. 

[image]

kylecorry31, 2025-09-01:
> Should we have an optional qibla direction in compass? It would come in handy for prayer purposes and improve general navigation.

I agree with @alphafalcon,  I would recommend using a beacon for this. Thank you for providing the detailed instructions

kylecorry31, 2025-09-05:
> I think a magnifier feature would be a great addition! It would really help with accessibility and make it easier to inspect small details. Something like this: https://play.google.com/store/apps/details?id=com.sunfire.magnifyingglass

Thank you for the suggestion, here's an issue to track it: https://github.com/kylecorry31/Trail-Sense/issues/3122

endarthur, 2025-09-07:
hi! Thanks for the awesome app!

I'm not sure if this should be sent here or to the comms plugin repo, but it would be nice to be able to get temperature (and maybe humidity and pressure) information from bluetooth devices. Maybe from devices using [bthome](https://bthome.io/), or some other seemingly more open devices, like ruuvitag. It might be useful for more accurate information than the phone's sensors, and it could always fallback to the phone if the connection is lost.

kylecorry31, 2025-09-07:
> hi! Thanks for the awesome app!
>
> I'm not sure if this should be sent here or to the comms plugin repo, but it would be nice to be able to get temperature (and maybe humidity and pressure) [...]

Thank you for the suggestion, I'm tracking plugin information here: https://github.com/kylecorry31/Trail-Sense/issues/1540 - that includes looking into Gadgetbridge and other alternatives for communicating with bluetooth devices such as thermometers.

kylecorry31, 2025-09-08:
> Some small additions for working "semi-manually" with maps:
> 
> * draw a line along a bearing from the current position or a beacon (or towards it)
> * showing distance and bearing directly when selecting a point on the map (or is there a reason to 'hide' that behind the navigate-button?)

Hi, thank you for the suggestions.

1. Can you clarify when the line would be drawn? Right now there's a line that gets drawn when navigating to the destination (if you have the Navigation layer enabled)
2. I believe https://github.com/kylecorry31/Trail-Sense/issues/2972 will do what you are looking for there (allow selecting a point without navigating but showing the details of that location). I also updated https://github.com/kylecorry31/Trail-Sense/issues/3009 to include the distance/direction

ElonSatoshi, 2025-09-17:
If the phone doesn't have a working compass sensor, the compass view should automatically show where the sun and moon is supposed to be

[13 more replies]