kylecorry31/Trail-Sense #2592: System black theme

kylecorry31, 2024-10-21:
The same as the system theme, except it uses the black mode instead of dark.

kylecorry31, 2024-10-21:
If possible, detect if "amoled black" is enabled

7eventech77, 2024-10-22:
I was going to actually suggest this idea myself without seeing the other thread shortly. lol.

Only thing I would suggest that would make it better is allow the user to select what color they want for both the "light" and "dark" modes of trail sense that will activate when set to "system color" mode. Different people may prefer a different color for "dark" mode from the current three modes night, dark, and black. I personally prefer black. For light mode white is okay for me but others may want a different color. This would probably be the best way to implement this and be an "be all end all" for this issue instead of selectively choosing the color based on if a device has an AMOLED screen which restricts everyone else who doesn't have an AMOLED screen from using this feature. (which may not be the easiest thing to detect).

kylecorry31, 2024-10-22:
> I was going to actually suggest this idea myself without seeing the other thread shortly. lol.
>
> Only thing I would suggest that would make it better is allow the user to select what color they want [...]

So maybe an entirely custom theme:

- Trigger (system, sunrise/sunset)
- System light / sun up (user can choose dark, light, black, night)
- System dark / sun down (user can choose dark, light, black, night)

kylecorry31, 2024-10-22:
Also, for reference the AMOLED black detection wouldn't be me detecting if the device has AMOLED, but rather the system setting for using AMOLED black that was posted in the new feature thread.

7eventech77, 2024-10-22:
> So maybe an entirely custom theme:
>
> * Trigger (system, sunrise/sunset)
> * System light / sun up (user can choose dark, light, black, night)
> * System dark / sun down (user can choose dark, light, [...]

7eventech77, 2024-10-22:
Also, adding a trigger "static" or similar label so that the color doesn't change ever except for direct user input would be good to have as well if the user wants to change the theme manually. Almost completely forgot!

kylecorry31, 2024-10-22:
> Also, adding a trigger "static" or similar label so that the color doesn't change ever except for direct user input would be good to have as well if the user wants to change the theme manually. Almost completely forgot!

I believe the static would just be the existing themes in Trail Sense, right?

7eventech77, 2024-10-22:
> > Also, adding a trigger "static" or similar label so that the color doesn't change ever except for direct user input would be good to have as well if the user wants to change the theme manually. Almost completely forgot!
> 
> I believe the static would just be the existing themes in Trail Sense, right?

Yes, that would be correct

kylecorry31, 2024-10-23:
Moving the custom theme over to this issue to keep this focused on just the system black theme: https://github.com/kylecorry31/Trail-Sense/issues/2596

PR #2598 (2024-10-26): Add system black theme
Closes #2592
closed 2024-10-26 by kylecorry31