kylecorry31/Trail-Sense #2595: Research brouter app

kylecorry31, 2024-10-23:
Questions to answer:
- [ ] Is it worth integrating with the Android app?
- [ ] What benefits will be provided by integrating?
- [ ] How likely are Trail Sense users to also use brouter?
- [ ] What does the integration look like?
- [ ] How much effort is the integration?
- [ ] Can I take inspiration instead for new features rather than delegating?
- [ ] Is this better to just recommend as a companion app?

damianofalcioni, 2024-10-24:
Some Hiking app integrating [brouter](https://github.com/abrensch/brouter) that users of Trail-Sense may already use:
- [OsmAnd](https://github.com/osmandapp/OsmAnd) : One of the most complete and configurable offline map and navigation app. (open source) 
- [OruxMaps](https://www.oruxmaps.com/): navigation app focused on outdoors activities (proprietary, free)
- [Locus](https://www.locusmap.app/): another navigation app focused on hiking, biking, or geocaching (proprietary, free)
- [AlpineQuest](https://alpinequest.net/) : my personal favorite hiking app. Lots of features, light and battery efficient (use multiple raster tile maps provider instead of vector maps), and offline (proprietary, paid version)
- [All-In-One Offline Maps](https://www.offline-maps.net/) : free version of AlpineQuest (proprietary, free)
- [Cruiser](https://github.com/devemux86/cruiser): offline navigation app (open source)
- [cgeo](https://github.com/cgeo/cgeo): famous Android geocaching app (open source)

kylecorry31, 2024-11-24:
@damianofalcioni thank you for the research!

michaelblyons, 2024-11-24:
Don't the data files for brouter take a ton of space? If you already have them downloaded for a different reason, I guess that's great.

damianofalcioni, 2024-11-24:
> Don't the data files for brouter take a ton of space? If you already have them downloaded for a different reason, I guess that's great.

Yes, as you see in my previous posts there are some famous hiking apps integrating brouter, that Trail-Sense users may potentially have and compared to osm or other vectorial maps they are really light! (consider the full Grece is 40mb, but on average is around 100mb for an area of 5*5-degree square).

damianofalcioni, 2024-11-26:
What does the integration look like:
Here the main code used by OSMAnd for the integration of brouter:

- Interface [IBRouterService.aidl](https://github.com/osmandapp/OsmAnd/blob/f0c87cfd7a0cc067f45b539a69ff50009fe3a1d9/OsmAnd/src/btools/routingapp/IBRouterService.aidl)
- binding [BRouterServiceConnection.java](https://github.com/osmandapp/OsmAnd/blob/f0c87cfd7a0cc067f45b539a69ff50009fe3a1d9/OsmAnd/src/btools/routingapp/BRouterServiceConnection.java)
- usage function returning the gpx or kml [findBROUTERRoute](https://github.com/osmandapp/OsmAnd/blob/f0c87cfd7a0cc067f45b539a69ff50009fe3a1d9/OsmAnd/src/net/osmand/plus/routing/RouteProvider.java#L1249)

Hope this help

kylecorry31, 2024-12-27:
I am not planning to add this myself - but if someone would like to implement this feature and submit a PR I will likely accept it. Just be sure to answer the questions in the description of this issue and demonstrate it working (preferably add it to the user guide).