kylecorry31/Trail-Sense #2659: More potential useful units for conversion tool

7eventech77, 2024-11-13:
Temperature - Kelvin
Weight - Milligrams
Time - week, month, year
Area - sq2 ft, sq2 meter etc.

Feel free to add what you think may be useful

JensKorte, 2025-02-06:
Battery types to measures and reverse. E.g. AA from Wikipedia: An AA cell measures 49.5–50.5 mm (1.95–1.99 in) in length, including the button [terminal](https://en.m.wikipedia.org/wiki/Battery_terminal)—and 13.7–14.5 mm (0.54–0.57 in) in diameter.

kylecorry31, 2025-02-06:
> Battery types to measures and reverse. E.g. AA from Wikipedia: An AA cell measures 49.5–50.5 mm (1.95–1.99 in) in length, including the button [terminal](https://en.m.wikipedia.org/wiki/Battery_terminal)—and 13.7–14.5 mm (0.54–0.57 in) in diameter.

Thank you for the suggestion. Can you clarify what this would be used for?

JensKorte, 2025-02-06:
Good question, I cannot answer it.