kylecorry31/Trail-Sense #2733: Suggestion: Add what3word as a tool

nebula-it, 2024-12-14:
Hey @kylecorry31,

Thanks for this amazing project. I've found all the little tools very helpful.
Is there a chance you can include https://github.com/what3words/w3w-android-wrapper when you have time. Merci

kylecorry31, 2024-12-14:
Hi @nebula-it, thank you for the suggestion.

I have an issue to track additional coordinate formats (I see someone recommended not using what3words), here:  https://github.com/kylecorry31/Trail-Sense/issues/785

In that issue linked above, would you mind providing the following details so I can determine if it's possible to include this and if it is a right fit for Trail Sense? 

1. Is there a version of this that doesn't require the Internet?
2. How widespread is the use of what3words?
3. How beneficial is this to hiking, backpacking, camping, and geocaching?
4. Why would I use this format over another coordinate format?
5. It was mentioned in that linked issue that it is a proprietary format. Can you confirm this and if so are there open source alternatives?

closed 2024-12-14 by kylecorry31

WalkTheEarth, 2025-01-30:
this isnt even good for rescues in mountains. no. use something like the OSGB if your in scotland.

kylecorry31, 2025-01-31:
More context: https://github.com/kylecorry31/Trail-Sense/issues/2801#issuecomment-2626590162