kylecorry31/Trail-Sense #2739: Add church/urban icons to beacons

LeftyDextrous, 2024-12-18:
Hello,

I was beacons for asylums and other safe places to go in the event I get lost or am in a vulnerable situation. I know of several churches/religious places I can trust to take care of/protect me or others in an emergency. It's not terribly likely I'll need them, but it's nice to have. For the sake of organization I would suggest adding a few icons for churches, safe houses, and any other places of asylum you can think of.

Also worth considering are icons for strictly urban locations and points of interest. I can't think of any at the moment but it's clear the current set available was curated with hikers in mind. However, this app may still prove useful in an urban environment, albeit only in the event of major infrastructure collapse or user paranoia. Still, I would argue either of those are within the spirit of the app, which I contend is for navigation and basic survival without internet (although so far this has been in the context of a rural/wilderness environment, and the "Use Cases" section reflects this bias).

Use case: Any scenario where you need to or must navigate an urban area without an internet connection (and this may happen more than you think).

Thank you for continuing to develop this app. Consider people like me who have found or may find this app helpful even in urban situations!

kylecorry31, 2024-12-18:
Thank you, I will look into adding additional icons, though Trail Sense will remain focused on hiking, backpacking, camping, and geocaching (as you found in the use cases section). I don't want to add too many non-wilderness icons, but will allow a few (3-4) as long as they could potentially have a wilderness use case. I will eventually add support for custom icons with #2740 which I believe will meet what you are looking for.

If you could provide links with recommended icons you would like included, that would be great. I normally get them from pictogrammers or the US national park service. Preferably for the church a religious agnostic one (or something that could represent a sanctuary).

kylecorry31, 2024-12-27:
After more consideration, I am not going to add these directly to Trail Sense, but rather allow users to add custom icons with #2740 instead. This allows me to keep Trail Sense focused on its core use cases while also allowing you to add urban icons as you see fit.