kylecorry31/Trail-Sense #2765: Augmented reality maximum view distance

astewart77, 2025-01-09:
Why is the maximum allowed only 100 feet? If I am following a backtrack, but not constantly watching my phone, it's easy to walk more than 100 feet past a turn. When I suspect I missed it and turn around, I can't see the path.

Also, when looking to cross a path from a gpx file with points more than 200 feet apart, I walked between two points without seeing the path when I crossed it. It took an arbitrary walk to the right or left at a probable intersection to pick up the path.

Is there a computational difficulty with a larger maximum?

Alan

kylecorry31, 2025-01-09:
Hi, the maximum path view distance was set due to performance concerns based on real world testing. I can experiment with increasing it, with a warning that it may degrade performance.

Trail Sense uses the following algorithm, so it should be able to handle spaced out points in most cases:
- Interpolate points on the path to obtain the desired spacing (this should handle spaced out paths)
- Snap to the nearest point if within ~60 feet (accounts for most GPS drift)
- Clip to the bounds
- Render the points to the screen

kylecorry31, 2025-01-21:
Will be uncapped in 6.8.0

closed 2025-01-21 by kylecorry31