kylecorry31/Trail-Sense #2767: Show angle of center bubble level

kylecorry31, 2025-01-09:
Draw the angle label on the center bubble level. Look into displaying this in the title as well (just call it angle?).

michaelblyons, 2025-01-10:
`r` for radius?

kylecorry31, 2025-01-10:
> `r` for radius? Theta for angle?

I'm just a bit concerned that those are too mathematical (X and Y are already kind of there).

darrencoen, 2025-01-10:
If i can chime in, it probably is Theta θ because it is the resulting vector of the X and Y. 

If i place my phone on an angled surface, if i rotate my phones orientation but keep it on the surface, the X and Y will change based on where the top of the phone is oriented. But the Theta will always remain the same as it is a combination of the two vectors and how they combine to coordinate the bubble from the origin.

[image]
[image]
[image]

[image]

michaelblyons, 2025-01-10:
https://en.m.wikipedia.org/wiki/Polar_coordinate_system

kylecorry31, 2025-01-10:
I think I will most likely just go with calling it "angle" since that's a common term. I feel like theta and vector may be too mathematical.