kylecorry31/Trail-Sense #2782: Why do I need to manually restart background services?

yihanwu1024, 2025-01-18:
I think this is a bug.

[image]

I use Backtrack from a Quick Settings tile. More often than not it results in this notification: "Restart services Background services could not be restarted. Tap to restart."

This should not be necessary if the user has already set Trail Sense to Unrestricted background usage.

kylecorry31, 2025-01-18:
That does sound like a bug, what Android version are you using? And I assume this happens when you click the notification?

(There's a big in android around quick settings tiles, but my workaround should cover that)

yihanwu1024, 2025-01-21:
I just replicated this on a Pixel 7 with Android 15.

kylecorry31, 2025-01-21:
@yihanwu1024 is there a set of steps you are taking to replicate this issue?

kylecorry31, 2025-01-21:
Thank you, I will look into this further

yihanwu1024, 2025-01-22:
I think I made an error in the previous comment. I must have falsely believed that I granted always-on location, in step 0. Below is new, accurate information.

Now, I can only reproduce the bug by setting Location access to "Allow only while using the app". Other steps remain the same.

Regardless, the error message in this situation is not helpful.

I also discovered a potential complication of this bug:

0. Install app and grant foreground precise location and unrestricted background usage.
1. Install the backtrack tile.
2. Launch Trail Sense, and go home right after.
3. Activate the tile, which will fail. (You would think this is the same as the original bug report. But:) Different from the original bug report, this time no notification is sent.
4. Launch Trail Sense from launcher. Observe that Trail Sense has crashed.

kylecorry31, 2025-01-22:
Thank you for the details, I'm able to reproduce the bug now

kylecorry31, 2025-01-22:
Unfortunately it doesn't look like the workaround works anymore on Android 15. This was put in place for a known issue in the Android operating system: https://issuetracker.google.com/issues/299506164, but it has regressed even further in Android 15. 

I'll do what I can to make sure it won't crash, but without background location it will show the restart services notification.