kylecorry31/Trail-Sense #2793: Enter multiple images for a single field guide page

kylecorry31, 2025-01-25:
- [ ] Allow the entry of multiple images
- [ ] Display multiple images (pager)
- [ ] Let user set the preferred image?

MikaelaLovett, 2025-04-27:
I absolutely adore the field guide! 
This option would be amazing as I want to add my drawings to keep the aesthetics of the default items, but I'd love to be able to see what the plants actually look like (photos) without downloading offline pages on Wikipedia. ^^ This is so exciting, I'm keeping such a guide physically and this is even better 🌿

kylecorry31, 2025-04-27:
Thanks! I'll likely implement this in the next couple of months, maybe a little sooner (I want to add animal track drawings to a couple of built in entries).

MikaelaLovett, 2025-04-27:
> > I absolutely adore the field guide! This option would be amazing as I want to add my drawings to keep the aesthetics of the default items, but I'd love to be able to see what the plants actually look like (photos) without downloading offline pages on Wikipedia. ^^ This is so exciting, I'm keeping such a guide physically and this is even better 🌿
> 
> Thanks! I'll likely implement this in the next couple of months, maybe a little sooner (I want to add animal track drawings to a couple of built in entries).

Thank you so much, I'll be looking forward to it! I'm now wondering, could you please add the option to edit the existing entries as well? (Some time in the future) ✨

I have a template with categories (size, colour etc.) that I use and it'd be lovely to add that description there. 😌

kylecorry31, 2025-04-27:
> Thank you so much, I'll be looking forward to it! I'm now wondering, could you please add the option to edit the existing entries as well? (Some time in the future) ✨
>
> I have a template with [...]

I'll think about allowing an edit to the built-in guide, but it may be easier to go with the functionality of "deleting" the built-in guide page and writing your own (#2881) (easier from a development perspective - since I would have to merge changes to the built-in pages with the user edits)

MikaelaLovett, 2025-04-28:
> I'll think about allowing an edit to the built-in guide, but it may be easier to go with the functionality of "deleting" the built-in guide page and writing your own (#2881) (easier from a [...]

Oh I didn't even think of that! The only shame is that it would delete your drawings as well, but I could redraw them in a similar style myself ^^ Thank you!

closed 2026-08-19 by kylecorry31