kylecorry31/Trail-Sense #2839: Some parts of the app are in another language, some are in english

daniele-athome, 2025-03-02:
I have a Pixel 8 running Android 15 with the following system languages configured in this order (on Android 15 you can configure more than one and give them priorities):

* English (UK)
* Italian

Since the latest update a couple of weeks ago, I noticed the sunset alarm notification was in Italian. Almost all the app is in English, however, beside that notification, a few strings are in Italian:

[image]

[image]

The only way to make it right is to remove Italian from the system languages - but I'd rather not do that because that setting also controls which languages my input methods make available.

kylecorry31, 2025-03-03:
Hi, I believe this is intended behavior of Android itself. This cause is that the string(s) you are seeing in Italian  are not present in the UK English translations.

You can get around this by:
1. Contributing the missing translations for UK English:  https://github.com/kylecorry31/Trail-Sense/issues/213. Requires a new release (later this month)
2. Setting the per app language on Android 15 (I believe under app info in system settings)
3. Adding US English as a preference after UK English but before Italian
4. This issue (open to community contribution): https://github.com/kylecorry31/Trail-Sense/issues/403

kylecorry31, 2025-03-04:
I'm going to close this in factor of #403 / #213

daniele-athome, 2025-03-04:
Sorry, I was doing some tests and I had to wait for the sunset notification alarm which is triggered once a day eheh (too lazy to do tests from Android Studio :-)

I thought Android would fall back to the "nearest" available language in the app (en-GB -> en), or the app default language at least (which is English if I'm not mistaken). It's been some time since I used to develop on Android so something probably has changed over the years.

I'll have more news about all my tests tonight.

daniele-athome, 2025-03-04:
Ok so:

* Solution 3 works
* Removing Italian from system languages works
* Setting English (US) as app-specific language in system settings didn't work (this should probably be a bug, shouldn't it? But whose bug, Android or Trail Sense?): both sunset notification and the screens reported in the original report were affected

According to [Android documentation](https://developer.android.com/guide/topics/resources/multilingual-support#resource-resolution-examples) - if I'm getting this right - the system will not even consider en-US when there is Italian in the list after en-GB. I'm guessing the "right" fix here would be to add en-US before Italian, as you proposed... however the app-specific language setting should have worked IMHO.

daniele-athome, 2025-03-05:
Another note about this: I had to also set app-specific language to en-US to make it work. Adding en-US after en-GB in system settings wasn't enough.

I'm really confused... I'll do the test app and let you know.

kylecorry31, 2025-03-05:
Thank you, much appreciated!

I wonder if Android requires both the language available at the system level and app level for that to work (that would be odd, but I could see that being the case)

kylecorry31, 2025-05-23:
It looks like there's a similiar known issue on Samsung devices where having 2 system languages causes the app to default to the wrong language (I had a user report this on Google Play and found a couple of open Samsung tickets).

daniele-athome, 2025-05-23:
Unfortunately I didn't find the time to produce a PoC to submit to AOSP issue tracker. On Trail Sense in my phone, the behavior is pretty much random: sometimes I have Italian, sometimes English, I wasn't able to find a pattern.

kylecorry31, 2025-08-11:
> More or less useful update: upgrading to Android 16 fixed the issue, so I'm guessing it was an upstream problem.

Thanks for the update! I found an issue logged for this on either the Google issue tracker or Samsung's tracker, unfortunately I never added it to this issue. But it did appear to be an upstream issue. Since it is fixed for you (and I assume other users who upgraded), I'm going to close this out.

closed 2025-08-11 by kylecorry31