kylecorry31/Trail-Sense #2877: Consider markdown file extension for guide pages

michaelblyons, 2025-04-04:
Your guide appears (from my spot checks) to be composed in Markdown. Is there a reason you are using the `.txt` file extension instead of `.md`?

The benefits are admittedly small, but GitHub will render pages by default and text editors have more of a chance at setting the right syntax highlighting by default.

kylecorry31, 2025-04-04:
Yes, the reason is that weblate (translations) only supports .txt files for translatable resources like that. It's annoying (VS Code never remembers that I want it highlighed as markdown), but hopefully weblate will support .md files in the future.

michaelblyons, 2025-04-04:
That makes sense, thanks! Two more thoughts:

 1. Change the close type of this issue as Not Planned in the drop down below.
 2. I think there's a way to make vscode project settings choose a particular syntax based on path rules as well as just the file extension. (Just if it annoys you. I mostly use Sublime Text, which does some of that differently.)

closed 2025-04-04 by kylecorry31