kylecorry31/Trail-Sense #2913: Convert tool quick action

kylecorry31, 2025-04-26:
Quick action to open the convert tool in a bottom sheet for quick conversions.

tosinonikute, 2025-05-19:
Hi @kylecorry31 Can I volunteer to work on this issue? I forked the project & did some work.

tosinonikute, 2025-05-19:
> > Hi [@kylecorry31](https://github.com/kylecorry31) Can I volunteer to work on this issue? I forked the project & did some work.
> 
> Thank you for the offer, that would be great - let me know if you need any clarifications. I was just thinking of having the entire convert tool fragment appear in a bottom sheet, but if you find something else that has a better UX, I'm open to suggestions.

Showing the entire convert tool in a bottom sheet is a good idea. For better user experience, I suggest displaying just 3 main convert tools in the bottom sheet tab. Users can then long-press the convert icon under QuickActions to reveal all the additional options on the main Convert Tool Screen.

See video:
https://github.com/user-attachments/assets/c29fe2e4-6280-41ad-842c-fad851b03455

kylecorry31, 2025-05-19:
> Showing the entire convert tool in a bottom sheet is a good idea. For better user experience, I suggest displaying just 3 main convert tools in the bottom sheet tab. Users can then long-press the [...]

That's a good idea, that look great!

tosinonikute, 2025-05-20:
> That's a good idea, that look great!

Cool, PR here: https://github.com/kylecorry31/Trail-Sense/pull/2952

closed 2025-05-20 by kylecorry31