kylecorry31/Trail-Sense #2918: Survival guide search quick action

kylecorry31, 2025-04-27:
A quick action to search the survival guide and see the relevant section. It will display the section and a button to open it in the survival guide.

tosinonikute, 2025-05-20:
Hi @kylecorry31 can I help with this?
Although I would need some clarifications.

kylecorry31, 2025-05-20:
> Hi [@kylecorry31](https://github.com/kylecorry31) can I help with this? Although I would need some clarifications.

Sure, I hadn't planned for this one to be open to community contribution so I will add additional details later today (I haven't fully thought out the design). If you have questions feel free to post them here and I'll answer them after as well.

tosinonikute, 2025-05-20:
> Sure, I hadn't planned for this one to be open to community contribution so I will add additional details later today (I haven't fully thought out the design). If you have questions feel free to [...]

Okay, my question is: The survival guide currently allows users to tap on search results section to navigate to the corresponding page. Given this existing functionality when survival guide is added on Quick Action, would adding a separate button to open the section in the survival guide be redundant or still useful?

kylecorry31, 2025-05-20:
Here's some more detailed requirements, I believe that will answer your question as well.

1. Button in the bottom sheet header that opens the survival guide. If the user entered a search query, it should carry that along (new argument will be needed for that fragment)
2. The user can enter a query into the search. It should use the same search approach as the survival guide tool, but only keep the top result. If the query is blank, no content will appear (maybe a message that says to search).
3. The section title of the top search result is shown.
4. There's a button that the user can click to open the section in the survival guide (basically the same as clicking a searched for section on the survival guide tool). It doesn't need a background, I just did that for the drawing to make it stand out.
5. The full markdown contents of the section should be rendered (not the summary that is shown in the search). You will need to load the chapter's markdown and extract the corresponding section. If you need to extract helper methods, I recommend to keep that in TextUtils with the other markdown parsing utilities.
6. A badge for the chapter should be displayed above the section title, just like it does on the list items on the survival guide tool.

[image]

I think the user experience may be better if the section title and above remain fixed while the section text is scrollable (with a fixed height, so when you search it doesn't jump around).

tosinonikute, 2025-05-21:
@kylecorry31 Nice detailed breakdown :+1:, will work on this & submit a pull request upon completion.

Will reach out if I need any clarifications during the process.

kylecorry31, 2025-05-29:
> [@kylecorry31](https://github.com/kylecorry31) quick question, if we need to use the same search approach as the survival guide tool, that means we should do something similar to `useSearchResults` done in `FragmentToolSurvivalGuideList`, but then all of those functions exist only in `AndromedaFragment`, but the `BoundBottomSheetDialogFragment` does not in any way extend `AndromedaFragment`.
> 
> In this case, what do you suggest will be the best approach?

I've been meaning to set up the bottom sheet with reactive support. I'll try to get that in place today so that you can leverage the same hooks.

kylecorry31, 2025-06-04:
> [@kylecorry31](https://github.com/kylecorry31) Yes, I should create a PR tomorrow or next.

Great, thank you!

tosinonikute, 2025-06-05:
> > [@kylecorry31](https://github.com/kylecorry31) Yes, I should create a PR tomorrow or next.
> 
> Great, thank you!

I have the PR here: https://github.com/kylecorry31/Trail-Sense/pull/2981

Kindly check it out & let me know If I'm missing anything.

closed 2025-06-05 by kylecorry31