kylecorry31/Trail-Sense #2923: Publish APK signing certificate hash

stefankreutz, 2025-05-01:
Could you please publish the SHA256 hash of the certificate used to sign the GitHub APK? That would help verifying the APK using [AppVerifier](https://github.com/soupslurpr/AppVerifier).

The [Trail-Sense-6.10.0-github-release.apk](https://github.com/kylecorry31/Trail-Sense/releases/download/6.10.0/Trail-Sense-6.10.0-github-release.apk) I just downloaded was signed with `D3:B7:CB:BC:63:3F:7C:1B:E2:CA:A2:F7:5F:B0:72:EE:9B:40:FD:10:FA:AF:96:42:69:2B:27:C4:AF:5E:C4:A7`.

kylecorry31, 2025-05-01:
Hi, I will consider doing that at some point. I need to do more research

stefankreutz, 2025-05-01:
Thank you!

You can view the signing certificate hashes using [apksigner](https://developer.android.com/tools/apksigner), too.

    apksigner verify --print-certs <apk>

AppVerifier uses the SHA256 hash in the format I included above.

kylecorry31, 2025-05-01:
Thank you for the instructions! I've added it here: https://github.com/kylecorry31/Trail-Sense/discussions/2868#discussion-8129200

closed 2025-05-01 by kylecorry31