kylecorry31/Trail-Sense #3020: Crashing when importing a PDF of substantial size

StarTroop, 2025-06-24:
### Where did you download the app from?

F-droid

### What app version are you using?

6.11.0

### What Android version and device are you using?

LineageOS 22.2 (Android 15), Motorola one 5g ace

### Frequency of issue

Always with larger PDFs (threshold currently unknown)

### Reproduction steps

I have a large GeoTiff file which I've converted to geospatial PDF using gdal. I've experimented with various combinations of options for JPEG compression and tiling off/on, for this report I'll focus on a 58mb pdf with tiling enabled and JPEG compression at 100.

Whenever I import the PDF, after the app completed auto-calibrating it immediately crashes. After reopening the app, trying to open the map again results in a crash.

### Expected behavior

Maps should open properly.

### List of diagnostic failures

Nothing relevant in diagnostics.
Error log suggests not enough memory, but the same (and larger) PDFs work in other apps on the same phone.

### Stack trace

```
Trail Sense 6.11.0

Package: com.kylecorry.trail_sense
Build type: release
Android SDK: 35
Device: motorola kiev_retail (motorola one 5G ace)
Fragment: MapListFragment
Diagnostics: weather-monitor-disabled
Message: Failed to allocate a 24 byte allocation with 1597600 free bytes and 1560KB until OOM, target footprint 268435456, growth limit 268435456; giving up on allocation because <1% of heap free after GC.

java.lang.OutOfMemoryError: Failed to allocate a 24 byte allocation with 1597600 free bytes and 1560KB until OOM, target footprint 268435456, growth limit 268435456; giving up on allocation because <1% of heap free after GC.
	at i4.k.p(SourceFile:143)
	at i4.k.o(SourceFile:123)
	at i4.k.onDraw(SourceFile:159)
	at e6.n.onDraw(SourceFile:37)
	at android.view.View.draw(View.java:25060)
	at android.view.View.updateDisplayListIfDirty(View.java:23903)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at androidx.constraintlayout.widget.ConstraintLayout.dispatchDraw(SourceFile:29)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at androidx.fragment.app.FragmentContainerView.drawChild(SourceFile:31)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at androidx.fragment.app.FragmentContainerView.dispatchDraw(SourceFile:36)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at androidx.fragment.app.FragmentContainerView.drawChild(SourceFile:31)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at androidx.fragment.app.FragmentContainerView.dispatchDraw(SourceFile:36)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at androidx.fragment.app.FragmentContainerView.drawChild(SourceFile:31)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at androidx.fragment.app.FragmentContainerView.dispatchDraw(SourceFile:36)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at androidx.constraintlayout.widget.ConstraintLayout.dispatchDraw(SourceFile:29)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.drawChild(SourceFile:14)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at android.view.View.updateDisplayListIfDirty(View.java:23894)
	at android.view.View.draw(View.java:24791)
	at android.view.ViewGroup.drawChild(ViewGroup.java:4568)
	at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4324)
	at android.view.View.draw(View.java:25063)
	at com.android.internal.policy.DecorView.draw(DecorView.java:794)
	at android.view.View.updateDisplayListIfDirty(View.java:23903)
	at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:694)
	at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:700)
	at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:798)
	at android.view.ViewRootImpl.draw(ViewRootImpl.java:5761)
	at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:5413)
	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:4412)
	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:3009)
	at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:10514)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1566)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1575)
	at android.view.Choreographer.doCallbacks(Choreographer.java:1175)
	at android.view.Choreographer.doFrame(Choreographer.java:1104)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1549)
	at android.os.Handler.handleCallback(Handler.java:991)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8934)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:591)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)

```

kylecorry31, 2025-06-24:
Thank you for the details, I will look into this. I suspect I may need to reduce the number of simultaneous PDF tile loads or reduce their resolution based on device available memory. I also believe #1345 will help, since that brings tiling fully under my control (Photo Maps uses a 3rd party library for photo rendering).

If I make changes would you be willing to try out a nightly build? I can provide instructions when I have a potential fix.

Also, if you are comfortable doing so, would you be able to share the PDF (either here or at trailsense@protonmail.com)?

kylecorry31, 2025-07-05:
@StarTroop I've restricted the max tile size of PDFs which I believe should resolve this issue for you. I can restrict it further if needed.

Can you try out the nightly build and let me know? https://github.com/kylecorry31/Trail-Sense/actions/runs/16088786315

Nightly instructions if needed: https://github.com/kylecorry31/Trail-Sense/discussions/1940

StarTroop, 2025-07-05:
Unfortunately I'm still crashing. I've sent the error report via email.

kylecorry31, 2025-07-05:
> Unfortunately I'm still crashing. I've sent the error report via email.

Thanks, I think I see what is happening (the nightly has much better stack traces). Do you happen to know the width and height of the PDF (in inches, mm, points, or pixels)?

I have another nightly build that you can try, which restricts the maximum size of the PDF (not ideal, but I can increase that if it works - and all of this is temporary because #1345 is replacing this PDF tiling engine and won't have that restriction): https://github.com/kylecorry31/Trail-Sense/actions/runs/16092815854

I'm assuming that if it works, zooming in won't be much more detailed than the reduced resolution version - but it let's me know what the problem is and I can either modify the library (open source) to deal with that or find the best balance for the resolution until #1345 is in place.

kylecorry31, 2025-07-06:
> The bandaid solution sorta works. There's no more crashing, but it is a little slow to reopen the map every time (a few seconds), and ultimately it is still blurry, but it's much more usable now. Are you compressing the PDF down to the max limit every time it's opened?
> 
> The PDF size should be 17578 x 28126 pixels, same as the source TIFF, but GIMP seems to report it differently so I'm not sure if there's some weird arbitrary scaling stuff going on. My PDFs are also generated with gdal's default tilesize of 256 x 256 (I assume that's pixels, but the documentation isn't very clear). gdalinfo reports the DPI as 141.

That's good to hear! Unfortunately with that library I don't have much control, so it is reopening the PDF for every tile (and it doesn't use 256x256 tiles, it uses an aspect ratio matching the PDF and automatically chooses a tile size). And it also doesn't store the reduced size PDF, and renders it at a lower sampling size for every tile. I think the thing that hurts performance is that it's opening the file for each tile AND blocking before the next tile is loaded. Ideally, I would open it once and grab all the tiles I need, which I'll be able to do once I do the tiling myself. 

I'm still working out some edge cases with my custom tiling approach on images, and will likely apply the same approach to PDFs next month.

StarTroop, 2025-07-06:
Thanks for all your effort! I'm looking forward to trying your new implementation when its ready.

kylecorry31, 2025-07-06:
Since this is no longer crashing, I'm going to close this out with the known issue about not so great resolution which will be addressed by the approach/issues mentioned here.

closed 2025-07-06 by kylecorry31