kylecorry31/Trail-Sense #3051: Improvements to the signal finder tool

naoliv, 2025-07-11:
### Description

**Cell towers could be grouped**; for example, in the screenshot below, it’s very likely that the same tower provides 2G, 3G, and 4G signals. Instead of showing separate entries, the tool could display a single line per tower, summarizing the types of signals it offers (e.g., "2G/3G/4G").

[image]

**There could be a directional indicator for the towers**, similar to how c:geo shows directions for geocaches. This would help users locate the towers more easily and understand their relative direction.

[image]

**Signal data from towers could potentially be summarized** in a more user-friendly way. As it stands (see screenshot), it’s unclear how the current raw list of, say, 5G signals is useful to the user. It might help to provide a clearer summary or visual representation, instead of listing all signals individually.

[image]

Thanks!

### Use case

Improve and simplify tower identification by grouping signals, showing tower direction relative to the user, and summarizing signal data for easier interpretation.

naoliv, 2025-09-21:
On this topic, I was in a place where only two signal indicators were showing:

[image]

My phone is dual SIM, so I believe these two signals correspond to each SIM card.
If that’s the case, would it be possible to also display the carrier or the SIM number?
This would make it easier to identify which carrier’s antenna has the stronger signal or is closer to our location.

kylecorry31, 2025-09-21:
> On this topic, I was in a place where only two signal indicators were showing: [...]

I will look into that, I don't think the carrier is possible without extra permissions, but the SIM number might be.